kizniche/ttgo-tbeam-ttn-tracker #1: 'os_init_ex' was not declared in this scope

VladoPortos, 2019-05-24:
Tried to compile the sketch but ended with, any idea how to fix that ?:
`C:\Users\VladoPortos\Desktop\ttgo-tbeam-ttn-tracker\main\ttn.ino: In function 'bool ttn_setup()':

ttn:113:57: error: 'os_init_ex' was not declared in this scope

     return ( 1 == os_init_ex( (const void *) &lmic_pins ) );

                                                         ^

C:\Users\VladoPortos\Desktop\ttgo-tbeam-ttn-tracker\main\ttn.ino: In function 'void ttn_cnt(unsigned char)':

ttn:201:24: error: 'LMIC_setSeqnoUp' was not declared in this scope

     LMIC_setSeqnoUp(num);

                        ^

exit status 1
'os_init_ex' was not declared in this scope
`

kizniche, 2019-05-24:
Which LMIC library are you using?

VladoPortos, 2019-05-24:
ah yes, wrong library :) its compiled now.

closed 2019-05-24 by VladoPortos