kizniche/ttgo-tbeam-ttn-tracker #3: Will not transmit

sivaelid, 2019-08-23:
The script load correctly and get very nice display showing the full detail with time incrementing messages n and message are shown as queued  and then a few second later  say sent . I have tested the board with other TTN scripts and work correctly. Lora 1 connected to pin 33 and Lora 2 conencted to pin 32. I have tried configured with 915Mhz and 868mhz and looked for rf with spectrum analyser . It looks a great TTN Mapper display but can not see what else  I should consider / configure. reagrds 
 BTW  thanks for the great work.

kizniche, 2019-08-23:
I'm not exactly sure what the issue is. You say the display works and the TTN Mapper works? Can you please explain what problem you are having?

rwanrooy, 2019-08-24:
I have the same problem. Configured the sketch first with ABP keys on 868 but it does not transmit. Everything looks fine on the oled (message sent is displayed). TTN does not receive any data. I tried OTAA as well but that doesn't work also. The board is fine because if I load another sketch the data coming in the ttn console. The version of the t-beam is T22 V1.0 20190612.

rwanrooy, 2019-08-24:
It works now. I changed the lmic_project_config.h in the library folder C:\Users\xyz\Documents\Arduino\libraries\lmic\project_config\lmic_project_config.h. The tracker now joins the network and everything works well :)

kizniche, 2019-08-24:
Thanks for the feedback. I'll include this in the README.

sivaelid, 2019-08-25:
Hi Thanks for the very fast response and hope  this helps  the later persons who had issues.
The hardware that I am using is the original T Beam (type 0) board that has been tested and used as a TTN mapper with other software.
The problem I had was that even though the software compiled correctly and worked as far as the display on the T beam   showed with all details correct. The display showed the message queued and then sent  the Lora module did not send ( I checked with RF spectrum analyser  and With my gateways) . I spent yesterday reviewing the code and my hardware and got it working Great.  Though at present I have not yet turned off the LoRa debug messages that I set. Change I made were as follows I use the 868MHz Eu region:
•	I made a clean install of the cade and the library’s particularly the LMIC which I had a couple of early ones very important.
•	 I set the library lmic file -  Libaries > ardunio-lmic-master > src>lmic> config set line 87 to  “#define LMIC_DEBUG_LEVEL 2” so that  can see all Debug messages in the serial console
•	In lmic file -  Libaries > ardunio-lmic-master > project_config>lmic_project_config> I set the  default  to  “define CFG_eu868 1” and commented out “//#define CFG_us915 1”. Should not be needed as set by the Ardunio IDE settings in lmic_project_config.h (see below)
•	In the Ardunio IDE I set the lmic_project_config.h so that “//#define CFG_us915 1 was commented out and “#define CFG_eu868 1” was uncommented
•	In configuration.h Tab I changed #define RESET_GPIO    from “23” to “LMIC_UNUSED_PIN” though this should not be needed as it set with in the ttn tab

This I found that this worked but many of the step may not be needed. I  will turn of Debug messages off, when do a final build. 
I would also not that I have found that with the SSD1306 displays I am using ( have tried with 2 different ones that the screen does not start cleanly and need to do a reset several times to bring up display and the time  to bring up the info screen is variable  and some times  need to reset again. May be just my hardware but have yet to look at the problem. 
May I say big thanks to Kyle for his work as the added display to The TTN mapper and newer Lmic library a great help. Many Thanks.

 Simon

kizniche, 2019-08-25:
I just updated the README with more setup details and my first attempt at getting this code to run on the new Rev1 T-Beam board. I have one tester with the board and I have one of these boards arriving in a few weeks. I'm closing this issue as it looks like both of you have everything working now. Thanks again for the help.

kizniche, 2019-08-25:
@rwanrooy If you want to test with your Rev1 board, edit configuration.h and comment T_BEAM_V07 and uncomment T_BEAM_V10. I think the code is now at a stable place to test the newer board.

rwanrooy, 2019-08-26:
I uploaded the sketch and can confirm it works fine. There is only one error after compiling the sketch:
C:\Users\roel\Documents\Arduino\libraries\lmic\src\hal\getpinmap_thisboard.cpp: In function 'const Arduino_LMIC::HalPinmap_t* Arduino_LMIC::GetPinmap_ThisBoard()':

C:\Users\roel\Documents\Arduino\libraries\lmic\src\hal\getpinmap_thisboard.cpp:61:72: note: #pragma message: Board not supported -- use an explicit pinmap

         #pragma message("Board not supported -- use an explicit pinmap")

                                                                        ^

If you compile the sketch again the error dissapears and uploads fine to the board. I had to remove the sats from the sketch. Do you know how to solve this error?

The new board has 3 buttons. Any toughts on how to get these buttons to work? Only the reset button works now. Maybe it's a good idea to assign a power off/on to the left button and a screen on/off to the middle one?

mtbspace, 2019-08-27:
> • In lmic file - Libaries > ardunio-lmic-master > project_config>lmic_project_config> I set the default to “define CFG_eu868 1” and commented out “//#define CFG_us915 1”. Should not be needed as set by the Ardunio IDE settings in lmic_project_config.h (see below)
> • In the Ardunio IDE I set the lmic_project_config.h so that “//#define CFG_us915 1 was commented out and “#define CFG_eu868 1” was uncommented
> 
> Simon

I found I had to do the same

I also had to select 'T-Beam' as the board in the Arduino IDE.  My T-Beam would keep rebooting if I selected 'Heltec_WIFI_LoRa_32' as the board

rwanrooy, 2019-08-28:
I can confirm the sketch works good on the v1.0 board. It's not rebooting if I use the T-BEAM board in Arduino. In the meantime I made a 3D case for the new version because the antenna has been moved. Maybe you can refer to it in the readme? https://www.thingiverse.com/thing:3830711

sivaelid, 2019-08-29:
I just wanted to say  the code is now great and having tested it again on the  tt22v0.7  works well. I have one issue in the serial output the Lng: and Alt: are nearly the same and Hdop is High I would have expected Alt around 70m and Hdop Less than 10
 I would just comment that  it has  has a voltage divider connected across the battery  with 2x 100k resister connected to Pin 35 . ( not sure if this is the case for the v1.0) . This is so that you can measure the battery voltage. I find this is good to see the state of the battery  when in use as a tracker.

 Thanks Simon

 
[Screen print of serial output.docx](https://github.com/kizniche/ttgo-tbeam-ttn-tracker/files/3554569/Screen.print.of.serial.output.docx)

closed 2020-08-20 by kizniche