kizniche/ttgo-tbeam-ttn-tracker #6: Lora not joining

netmonk, 2019-10-12:
I freshly installed an compiled this code on my Tbeam rev10 (latest). 
I already run on this device paxcounter with success to join lora TTN. 
I run also the GW, so i can see what message are transiting from it. 

I put the appeui lsb, deveui lsb and appkey msb, i cannot see any join in the gateway. 
This is for EU version standard payload.

netmonk, 2019-10-12:
first, the LED_PIN is set to 14, but it is also the Gpio for lora RST on rev10 so i i changed it to unused GPIO, but still nothing goes out from the board. Also if you look at the datasheet, rev10 only DIO0, and DIO1 and DIO2 are not defined. So tried to change them to LMIC_UNUSED_PIN in ttn.ino but no success.

netmonk, 2019-10-12:
```
// -----------------------------------------------------------------------------
// LoRa SPI
// -----------------------------------------------------------------------------

#define SCK_GPIO        5
#define MISO_GPIO       19
#define MOSI_GPIO       27
#define NSS_GPIO        18
#define RESET_GPIO      14
#define DIO0_GPIO       26
#define DIO1_GPIO       33
#define DIO2_GPIO       32
```
first attempt to correct the pinout

kizniche, 2019-10-13:
I just got the rev10 board, but I haven't had a chance to plug it in and mess with it. Let me know if you find or fix anything in your investigation.

Internetwurm, 2019-10-27:
https://github.com/kizniche/ttgo-tbeam-ttn-tracker/issues/8

maybe you take a look here ;-)

micbuh, 2019-12-29:
if not set to single channel gateway,  LMIC_startJoining() will never be executed.

_LMIC_startJoining();_ should before _#ifdef SINGLE_CHANNEL_GATEWAY_.

ttn.ino

```
#elif defined(USE_OTAA)

  #ifdef SINGLE_CHANNEL_GATEWAY
  // Make LMiC initialize the default channels, choose a channel, and
  // schedule the OTAA join
  LMIC_startJoining();

  // LMiC will already have decided to send on one of the 3 default
  // channels; ensure it uses the one we want
  LMIC.txChnl = SINGLE_CHANNEL_GATEWAY;
  #endif
#endif

```

kizniche, 2019-12-30:
Thanks for the info. I'm going to try to get back into this in 2 weeks when I have more time and try to test and incorporate changes.

closed 2020-08-20 by kizniche