kizniche/ttgo-tbeam-ttn-tracker #8: axp192 LoRa ( sx1276 ) Voltage

Internetwurm, 2019-10-27:
Hello @all,

i just added the set Voltage reading to the serial.printf block to take a look what is configured.
I discovered that for Channel 2 ( DCDC2 sx1276 channel ) it was set to 1.250V ( 1250 raw value).
Setting it to 1.850V doesn´t make it connect to ttn.
But with purely underpowered it could make a difference. 
Maybe it will need full Voltage to send for more far distance.

[image]

The adjusted block:
      Serial.printf("DCDC1: %s\n", axp.isDCDC1Enable() ? "ENABLE" : "DISABLE");
      Serial.printf("DCDC1 Voltage: %d\n", axp.getDCDC1Voltage());
      Serial.printf("DCDC2: %s\n", axp.isDCDC2Enable() ? "ENABLE" : "DISABLE");
      Serial.printf("DCDC2 Voltage: %d\n", axp.getDCDC2Voltage());
      if(axp.getDCDC2Voltage() < 1800)
      {
        axp.setDCDC2Voltage(1850); //sx1276 needs 1.8 Volte at least, so something extra for suitable powering 
        delay(100); // not sure if it is really needed
        Serial.printf("NEW DCDC2 Voltage: %d\n", axp.getDCDC2Voltage());
      }
      
      Serial.printf("LDO2: %s\n", axp.isLDO2Enable() ? "ENABLE" : "DISABLE");
      Serial.printf("LDO2 Voltage: %d\n", axp.getLDO2Voltage());
      Serial.printf("LDO3: %s\n", axp.isLDO3Enable() ? "ENABLE" : "DISABLE");
      Serial.printf("LDO3 Voltage: %d\n", axp.getLDO3Voltage());
      Serial.printf("DCDC3: %s\n", axp.isDCDC3Enable() ? "ENABLE" : "DISABLE");
      Serial.printf("DCDC3 Voltage: %d\n", axp.getDCDC3Voltage());
      Serial.printf("Exten: %s\n", axp.isExtenEnable() ? "ENABLE" : "DISABLE");
      Serial.printf("Battery Voltage: %d\n", axp.getBattVoltage());
      Serial.printf("BAttery Percentage: %f\n", axp.getBattPercentage());
      Serial.printf("Battery Charge Current set: %d\n", axp.getSettingChargeCurrent());
      Serial.printf("Chip Temp: %d\n", axp.getTemp());
      Serial.printf("Chip TSTemp: %d\n", axp.getTSTemp());

Internetwurm, 2019-10-27:
ah yes i missed to make a proofing picture from voltage value before. 
Maybe it also differs between Models.
I am using the V1.0.

Internetwurm, 2019-10-27:
God damn!
[image]
Set it to 2.275V for LoRa and rechecked my keys

IT WORKED!

kizniche, 2019-10-27:
Excellent investigative work! Could you tell me the exact change you made for the fix so I can replicate, test, and update the code?

kizniche, 2019-10-27:
Is it ```axp.setDCDC2Voltage(2275);``` ?

Internetwurm, 2019-10-27:
Okay i try to reveal my way.

On configration.h changed values form default:

"PAYLOAD_FULL"
LORAWAN_CONFIRMED_EVERY  10
LORAWAN_SF DR_SF12  // I am 15km off the next gateway ;-)
LORAWAN_ADR 1

Except the last two values it shouldn´t change anything important.

On "credentials.h"

define USE_ABP
Fiddeling with endian keys several times, maybe i am too old. 

Nothing else here.
All other unchanged, just added +1 hour for GPS time in "gps.h".

In my opinion the most important change comes here for sx1276 module voltage supply.
On "main.h"

i added at
if (axp192_found)
{ ...
//first serial.printf block
//axp.setPoweOutPut statements
//second Serial.printf block
I added ( insetered ) the commands from first post.

...}
I used reference Document to reveal command to read and set values from the "AXP202X_Library" at "/src/axp20x.cpp".
Then i added them to just read the set values.
After first run i added the if query to adjust DCDC2 voltage to ( now ) maximum 2.275V ( 2275 ).

With ABP i was able to send at least one packet over 15km with default delivered antenna.
Very impressive thing.

For reference the Link to axp chip family document.

https://github.com/m5stack/M5-Schematic/blob/master/Core/AXP192%20Datasheet_v1.1_en_draft_2211.pdf

So far, as i changed to OTAA now it doesn´t seem to reconnect.
If i set sleep time and reset it starts up and go to sleep admideatly.
Maybe a wating loop or time is needed here.
Also it could take some time to hit a gateway with the send packets.

But as shown with ABP at least it was a proof that your sketch is working.
Maybe i have to give it a try with CLOCK_ERROR to get it reliable work for a time.

Thanks for your work.
Hopefully you will transfer this to be used with ESP-IDF soon? ;-)
Just joking.

Internetwurm, 2019-10-27:
[image]

netmonk, 2019-10-27:
Im far away from any gateway right now, so basically i wont be able to test anything before end of week

Internetwurm, 2019-11-08:
Hey Beamer Folks,

as it seems OTAA doesn´t work for me for this far distance.
I got no luc in Joining with OTAA.

I will try to keep up with that but for now i hang off from this until i get at least a one channel gateway here. 
Doesn´t make sense without it.
It is not reliable in this constellation caused by the huge distance.

closed 2019-11-08 by Internetwurm