kizniche/ttgo-tbeam-ttn-tracker #10: Spreading Factor Increases on OTAA with ADR off

tkerby, 2019-11-06:
I've noticed that over time the data rate/SF increases up to SF10 with ADR turned off (set to 0) in the config file. I'm using OTAA join and have tried LMIC 2.3.2 and 3.0.99 from the MCCI repositories. I've tried setting interrupts both on and off in LMIC to see if its timing related.

See image below from the gateway. The number of packets received before the data rate increases is variable but I see some rejoin failed messages on the screen and message discarded (presumably through airtime constraints) in operation.

[image]

tkerby, 2019-11-06:
My thought on this is that to receive link dead messages in the lcd, link checking must be enabled. Could this be causing the SF to increase even with ADR off?

tkerby, 2019-11-06:
I made a mod to turn off link checks but I'm not seeing an improvement and it still increases SF

void ttn_adr(bool enabled) {
    LMIC_setAdrMode(enabled);
    LMIC_setLinkCheckMode(!enabled);
}

ctandi, 2021-01-13:
@tkerby Have you found a fix by now? I'm running into the same issue. Thanks.

pooley92, 2021-02-13:
I am also experiencing this too. Changing the DR_SF has no effect and it cycles through the data rates. Seems like it’s ADR.

Migi5700, 2021-03-18:
Hi everyone,
I have the same problem here, the DR_SF goes up and down.
Changing the DR_SF in the configuration.h file has also no effect, it is always starting at SF7.
I'm not sure if this is ADR, because I never saw a downlink-message, but I dont know if you see the ADR-messages at the TTN traffic-console.
Anyone can help :-)
Greetings

ciphermaster, 2021-05-02:
Hi everyone.
Same issue here. I have tried everything but the SF remains on SF7. It does not matter if I have enabled ADR or Not.
I have also disabled LMIC_setLinkCheckMode(0) but no change. LMIC Library is on the latest version.

BjoernSch, 2021-08-21:
> I made a mod to turn off link checks but I'm not seeing an improvement and it still increases SF
> 
> void ttn_adr(bool enabled) {
> LMIC_setAdrMode(enabled);
> LMIC_setLinkCheckMode(!enabled);
> }

This enables link checking if ADR is disabled.

closed 2022-07-03 by kizniche

gougakos, 2023-01-29:
> Hi everyone. Same issue here. I have tried everything but the SF remains on SF7. It does not matter if I have enabled ADR or Not. I have also disabled LMIC_setLinkCheckMode(0) but no change. LMIC Library is on the latest version.

have you to fixed the stuck SF7 issue. No matter what i try it always sends only to sf7. I opened  [https://github.com/kizniche/ttgo-tbeam-ttn-tracker/issues/55](url) regarding this.

commit b5a904c0 (2022-07-03): Fix #10