kizniche/ttgo-tbeam-ttn-tracker #12: 433mHz

skyden-ua, 2020-01-18:
Hello. Can you add 433mHz? I tried to do this but have persistent problems.

kizniche, 2020-01-24:
Did you perform [step 3 of the README](https://github.com/kizniche/ttgo-tbeam-ttn-tracker#setup)?

> Copy the contents of the project file main/lmic_project_config.h to the library file arduino-lmic/project_config/lmic_project_config.h and uncomment the proper frequency for your region.

kizniche, 2020-01-24:
Oh, I see it's not in there. Let me look into this.

kizniche, 2020-01-24:
There doesn't appear to be support for 433 MHz in the arduino-lmic library, but [one user may have found how to add support](https://github.com/mcci-catena/arduino-lmic/issues/96#issuecomment-511898685).

closed 2020-08-20 by kizniche