kizniche/ttgo-tbeam-ttn-tracker #13: WIP for deep sleep support

geeksville, 2020-01-19:
Hi,

I just found your great project and I'm going to use it as a starting point on my LORA futzing.  Alas, I've just ordered my TBEAMs from china and begun refactoring to allow me to use the axp192 (and the ssd sleep command) to have the device sit in deep sleep most of the time.  My devices won't be here for a couple of weeks, but once they arrive and I test the changes I'll send you a PR.

However, if you are interested in trying deep sleep before then, here's my WIP branch: 
https://github.com/geeksville/ttgo-tbeam-ttn-tracker/commit/ddf47aa1342b1ab8987a4ba11b7c31f1c3e37b12

I had previously used other TTGO devices on a different (non LORA) project and exchanged emails with them about some poor choices they made wrt power rails and sleep.  It seems that this new device is much better (but still not perfect).

geeksville, 2020-02-01:
Alas, I still haven't received my TBEAM with a screen, so I'll wait to make a clean PR until I can test with that.  But I did find a TBEAM without the oled and it seems to work well.  Changes compared to your version:

* GPS and LORA radios powered down during deep sleep (so current draw during deep sleep is now much lower)
* OTAA joining works properly (resulting keys are stored in flash and device only has to join once)
* the lora packet count is stored in RTC ram, so it properly keeps going up after sleep.  (and very occasionally it is mirrored to flash - to guarantee we never fall behind what the TTN service expects it to be - even during development or if you kill the battery)
* a three sec  long press of the GPIO button means 'discard lora prefs and rejoin'

WIP here: https://github.com/geeksville/ttgo-tbeam-ttn-tracker/commits/master

kizniche, 2020-08-20:
Very nice work. Have you received your screen yet? I would welcome a PR.

kizniche, 2020-08-20:
I tested with the OLED and merged. One thing I tried to add that I couldn't figure out was getting the display to work after sleep. If you have any clue as to how to do this, please share. Thanks.

FabTangi, 2020-08-24:
Hello,
display->displayOn();
display->displayOn();
does not work on my TBeam V1.0. Screen remains off after deep-sleep...

geeksville, 2020-08-24:
alas ;-)