kizniche/ttgo-tbeam-ttn-tracker #15: cayenne mydevices device

chrkan, 2020-02-10:
Hallo Tanks for the great work, 
can you tell me witch device you use vor cayenne mydevices to show the track in a map?

thanks

kizniche, 2020-02-11:
I don't quite understand the question. Can you elaborate?

chrkan, 2020-02-12:
In your skatch there is the option PAYLOAD_USE_CAYENNE, my question is how you can configure this at https://cayenne.mydevices.com/ or which device you select there. Or did I misunderstand the whole thing?

kizniche, 2020-02-12:
This option is to use the Cayanne payload format for thethingsnetwork decoder rather than the supplied custom decoder.

[image]

closed 2020-08-20 by kizniche