kizniche/ttgo-tbeam-ttn-tracker #16: Replacing lmic_project_config.h overwrites new settings

AusUser5, 2020-03-05:
Hi there, I'm looking to use your sketch with AU915, the copy/paste instructions provided overwrite  latest version of the lmic_project_config.h as the lastest library version contains.

> // project-specific definitions
//#define CFG_eu868 1
#define CFG_us915 1
//#define CFG_au915 1
//#define CFG_as923 1
// #define LMIC_COUNTRY_CODE LMIC_COUNTRY_CODE_JP	/* for as923-JP */
//#define CFG_kr920 1
//#define CFG_in866 1
#define CFG_sx1276_radio 1
//#define LMIC_USE_INTERRUPTS<

Thought it may be better to take a copy form the original library source and edit as required.

kizniche, 2020-03-05:
Good suggestion. Their library config has changed since I wrote the install instructions. Perhaps it would be better to just instruct the user where this file is located and to edit it accordingly.

AusUser5, 2020-03-05:
Yes, that would be better I think. Thanks for sharing, great project.

closed 2020-08-20 by kizniche