kizniche/ttgo-tbeam-ttn-tracker #20: GPS doesn't work...

schroepfer, 2020-07-07:
... with t-beam Rev1.
My T_BEAM_V10 is defined in the configuration.h , the RX and TX of the GPS is set right.
The blue GPS-LED blinks, even the t-beam is switched off!

I searched GPS for hours. It doesn*t work. With Meshtastic GPS is working fine.
With the t-beam GPS-testsoftware from https://github.com/luckynrslevin/TTGO-T-Beam-GPS it also doesn't work.
Any hints?

Juergen

slavino, 2020-07-08:
double check your GPS RX PIN they moved it around with redesigns between the versions
the test code could be for older board

T22_V05 20180608 (https://www.banggood.com/LILYGO-TTGO-T-Beam-ESP32-433868915Mhz-WiFi-Wireless-bluetooth-Module-p-1320390.html)
T22_V06 20180705 (https://www.banggood.com/LILYGO-TTGO-T-Beam-433MHz470MHz868MHz915MHz-ESP32-WiFi-Wireless-bluetooth-Module-p-1345473.html)
T22_V07 20180711
T22_V1.0 20190612

schroepfer, 2020-07-08:
My board is T22_V1.0 20190612.
This is, what i set:

// Select which T-Beam board is being used. Only uncomment one.
//#define T_BEAM_V07  // AKA Rev0 (first board released)
#define T_BEAM_V10  // AKA Rev1 (second board released)
....
  #define GPS_RX_PIN      34
  #define GPS_TX_PIN      12

Conor-Burns, 2020-07-20:
I've got the same problem - meshtastic locks - this script does'nt
I did switch the gps pins around to no avail - both times the gps led started to flash after a while to indicate successful lock, but the software still stated no lock :/

schroepfer, 2020-07-20:
Yes, only Meshtastic works now for the t-beam 1.x . 
It seems, that ther is no tracker-software now for these boards.
hottimuc and noppingen tracker doesn't work too......

Conor-Burns, 2020-07-20:
@kizniche could you have a look at this and maybe compare it to meshtastics gps implementation?

kizniche, 2020-07-20:
I'll take a look at it.

mo-norant, 2020-07-30:
and any news?

hagenbuch, 2020-08-17:
I seem to have the same problem. I bought a T-Beam T22_V1.1 from Tindie. On the board, they are indicating 

`GPS TX: IO34 = ESP32 RX
GPS RX: IO12 = ESP32 TX`

Corresponds well to

`#define GPS_RX_PIN 34
#define GPS_TX_PIN 12`

The blue LED blinks, the red LED is rather dim. On the serial console, I can see this when booting:

`en:8896
load:0⸮⸮⸮⸮c
I2C device found at address 0x34 !
axp192 PMU found
done

AXP192 Begin PASS
DCDC1: ENABLE
DCDC2: ENABLE
LDO2: ENABLE
LDO3: ENABLE
DCDC3: ENABLE
Exten: ENABLE
----------------------------------------
DCDC1: ENABLE
DCDC2: ENABLE
LDO2: ENABLE
LDO3: ENABLE
DCDC3: ENABLE
Exten: ENABLE
TTN MAP-TRACK 1.1.1
TTN MAP-TRACK 1.1.1`

I also see "Charging - No Charging" when I press the button. A reset gives:

`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
I2C device found at address 0x34 !
axp192 PMU found
done

AXP192 Begin PASS
DCDC1: ENABLE
DCDC2: ENABLE
LDO2: ENABLE
LDO3: ENABLE
DCDC3: ENABLE
Exten: ENABLE
----------------------------------------
DCDC1: ENABLE
DCDC2: ENABLE
LDO2: ENABLE
LDO3: ENABLE
DCDC3: ENABLE
Exten: ENABLE
TTN MAP-TRACK 1.1.1
TTN MAP-TRACK 1.1.1`

kizniche, 2020-08-20:
I can confirm GPS works in the latest commit. I haven't been able to reproduce the issue. It also looks like you're no longer having this issue. Are we good to close this issue?

schroepfer, 2020-08-21:
.. i have still no GPS after hours. I use you newest release from today on my t-beam v1.0.

hagenbuch, 2020-08-21:
If you get no error when compiling and everything uploads fine the thing might be broken. Is the GPS antenna attached? Does a red LED close the the GPS chip blink? (Not the dim red one ).

Do you have a battery inserted (I know it's not necessary) and does the blue LED go on when charging? Do you see the starup messages in the serial console? Is your power source really good and stable?

schroepfer, 2020-08-21:
Hello hagenbuch,
if you read my comments, you wouldn't ask these questions!
(the module works fine with Meshtasic!)
The red LED is off, the blue one lights, when charging.

hagenbuch, 2020-08-21:
Sorry I didn't - then the question is, could Meshtastic (whatever that is) read the GPS?

schroepfer, 2020-08-21:
yes.
https://www.meshtastic.org/

hagenbuch, 2020-08-21:
Ah thanks.. looks great - well then I would try to switch on all available debugging signals, is LoRa working correctly? Could you create and transfer the AppEUI, DevEUI (both lsb) and AppKey (msb)? Does the TTN server show data?

schroepfer, 2020-08-21:
I didn't test anything else. I first wait, until GPS is working....

hagenbuch, 2020-08-21:
I'm sure you've set 

`#define T_BEAM_V10`

...

schroepfer, 2020-08-21:
sure! Aber 100%ig !

bigal488, 2020-08-21:
Just to add to the conversation on this.
We have 5 T-Beams all V1.1 20191212

ttgo-tbeam-ttn-tracker is running on all 5 devices and all have a GPS lock. However, I just yesterday received a new batch from Aliexpress (still v1.1 20201212 with Neo 6M) and the first one I’ve tried will not get a GPS lock. Its been left out all night with a clear view of the sky and still nothing. It is running exactly the same code as the 5 other devices with only the TTN credentials changed.
However, the red GPS LED is flashing (suggesting a GPS lock in hardware?) but the software still reports “waiting for gps lock”. I’m going to load up meshtastic and see if that gets GPS.
Will let you know what happens….

schroepfer, 2020-08-21:
i think too, that there are differeces between the versions 20190612 (mine), 20191212 and 20201212 in the GPS.
Who knows?

bigal488, 2020-08-24:
We've come to the conclusion that Meshtastic has "broken" the TBeams GPS. We took a know good T-Beam running ttgo-tbeam-ttn-tracker (GPS lock acquired within a few minutes). Then flashed Meshtastic 0.91. That acquired a GPS lock within a few minutes. Then we reflashed ttgo-tbeam-ttn-tracker - now NO GPS. 
We loaded up a basic GPS serial comms sketch and theres no output from the GPS serial port. I tied the battery short reset but that doesnt make a difference.
I've filed a bug report on the Meshtastic Github page:
https://github.com/meshtastic/Meshtastic-device/issues/334
Just received a shipment of 6 TBeams from LilyGo (all shipped with Meshtastic) and none of them will acquire a GPS lock.

kizniche, 2020-08-25:
TinyGPS is very limited in scope, so I've been experimenting with [SparkFun_Ublox_Arduino_Library](https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library). I've been able to switch the GPS to NMEA mode and selectively enable/disable NMEA sentences, with the settings persisting across reboots. However, it seems TinyGPS still isn't properly parsing any of the NMEA sentences I've tried individually enabling. I haven't had a lot of time to work on this, but it's progress. Here's the code I used for the T-Beam V1.0:

```c#
//https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library
//http://librarymanager/All#SparkFun_Ublox_GPS
#include <SparkFun_Ublox_Arduino_Library.h>
SFE_UBLOX_GPS myGPS;

void setup() {
  Serial.begin(115200);
  while (!Serial);  // Wait for user to open the terminal
  Serial.println("Connected to Serial");
  Serial1.begin(9600, SERIAL_8N1, 34, 12);

  do {
    if (myGPS.begin(Serial1)) {
      Serial.println("Connected to GPS");
      myGPS.setUART1Output(COM_TYPE_NMEA); //Set the UART port to output NMEA only
      myGPS.saveConfiguration(); //Save the current settings to flash and BBR
      Serial.println("GPS serial connected, output set to NEMA");
      myGPS.disableNMEAMessage(UBX_NMEA_GLL, COM_PORT_UART1);
      myGPS.disableNMEAMessage(UBX_NMEA_GSA, COM_PORT_UART1);
      myGPS.disableNMEAMessage(UBX_NMEA_GSV, COM_PORT_UART1);
      myGPS.disableNMEAMessage(UBX_NMEA_VTG, COM_PORT_UART1);
      myGPS.disableNMEAMessage(UBX_NMEA_RMC, COM_PORT_UART1);
      myGPS.enableNMEAMessage(UBX_NMEA_GGA, COM_PORT_UART1);
      Serial.println("Enabled/disabled NEMA sentences");
      break;
    }
    delay(1000);
  } while(1);
}

void loop() {
  if (Serial1.available()) {
    Serial.write(Serial1.read());  // print anything comes in from the GPS
  }
}
```

eriktheV-king, 2020-08-25:
Hello
I've modified a LilyGO sketch with SparkFun libraries to help you out to reset U-blox NEO GPS devices on TTGO T-beam T22- V1.0 and 1.1 
This sketch will bring back U-blox GPS N6M & N8M factory settings so that NMEA 9600 over the GPS serial output is enabled.
You can download this sketch on my GitHub: 
https://github.com/eriktheV-king/TTGO_T-beam_GPS-reset
It worked fine for my 5 T-beams... good luck !

bigal488, 2020-08-25:
Fantastic. Thanks for your help.

kizniche, 2020-08-25:
@eriktheV-king, I modified this exact same Sparkfun example file, in pretty much the same way, however it did not restore functionality after flashing the ttn-tracker code. Have you been able to successfully get a GPS lock with ttn-tracker following an execution of the code you link to?

kizniche, 2020-08-25:
I also notice your code sets and saves UBX mode rather than NMEA:

https://github.com/eriktheV-king/TTGO_T-beam_GPS-reset/blob/93001713e34cee721eff7e8c863567483bb0ab58/GPS-T22_v1.0-20190612_v1.1-20191212-resetGPS/GPS-T22_v1.0-20190612_v1.1-20191212-resetGPS.ino#L91

t0xa, 2020-08-25:
> TinyGPS is very limited in scope, so I've been experimenting with [SparkFun_Ublox_Arduino_Library](https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library). I've been able to switch the GPS to NMEA mode and selectively enable/disable NMEA sentences, with the settings persisting across reboots. However, it seems TinyGPS still isn't properly parsing any of the NMEA sentences I've tried individually enabling. I haven't had a lot of time to work on this, but it's progress. Here's the code I used for the T-Beam V1.0:
> 
> ```cs
> //https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library
> //http://librarymanager/All#SparkFun_Ublox_GPS
> #include <SparkFun_Ublox_Arduino_Library.h>
> SFE_UBLOX_GPS myGPS;
> 
> void setup() {
>   Serial.begin(115200);
>   while (!Serial);  // Wait for user to open the terminal
>   Serial.println("Connected to Serial");
>   Serial1.begin(9600, SERIAL_8N1, 34, 12);
> 
>   do {
>     if (myGPS.begin(Serial1)) {
>       Serial.println("Connected to GPS");
>       myGPS.setUART1Output(COM_TYPE_NMEA); //Set the UART port to output NMEA only
>       myGPS.saveConfiguration(); //Save the current settings to flash and BBR
>       Serial.println("GPS serial connected, output set to NMEA");
>       myGPS.disableNMEAMessage(UBX_NMEA_GLL, COM_PORT_UART1);
>       myGPS.disableNMEAMessage(UBX_NMEA_GSA, COM_PORT_UART1);
>       myGPS.disableNMEAMessage(UBX_NMEA_GSV, COM_PORT_UART1);
>       myGPS.disableNMEAMessage(UBX_NMEA_VTG, COM_PORT_UART1);
>       myGPS.disableNMEAMessage(UBX_NMEA_RMC, COM_PORT_UART1);
>       myGPS.enableNMEAMessage(UBX_NMEA_GGA, COM_PORT_UART1);
>       Serial.println("Enabled/disabled NMEA sentences");
>       break;
>     }
>     delay(1000);
>   } while(1);
> }
> 
> void loop() {
>   if (Serial1.available()) {
>     Serial.write(Serial1.read());  // print anything comes in from the GPS
>   }
> }
> ```

I can confirm that after running this code on `Meshtastic T-Beam Rev1 T22_V1.1 20191212` ttgo-tbeam-ttn-tracker started to report GPS coordinates.

eriktheV-king, 2020-08-25:
> Have you been able to successfully get a GPS lock with ttn-tracker following an execution of the code you link to?

> I also notice your code sets and saves UBX mode rather than NMEA:

Hello Kyle

My sketch wakes up the T-beam's powerchip as in the original LilyGO example, but in the loop I used example code that came with the SparkFun library. So I didn't actually change much, just combined the necessary lines from both and greyed out the unnecessary.

It polls the U-blox to see if its UART runs on 9600 or 38400 baud, then sets speed to 38400. I left the line of code setting UBX output (could have left that out, too), then save the config.
GPS UART comes back at 38400, then hard resets, then issues a factory restore which restores NMEA output and 9600 baud.

As a test, it polls the Ublox for the protocol version, then keeps outputting cordinates on the T22-V1.1 
On polling the GPS protocol version after u-blox factory restore, the V1.0 t-beam often restarted due to a 'guru meditation error'. 
But when I loaded the original sketch from LilyGO https://github.com/LilyGO/TTGO-T-Beam/tree/master/GPS-T22_v1.0-20190612 , all 'unbricked' t-beam GPS'es work fine: within a few seconds started outputting good cordinates, and within 20 seconds the gps LED shows a fix, too.
I also successfully tested Luckynrslevin's https://github.com/luckynrslevin/TTGO-T-Beam-GPS sketch.

I then tried your ttn-tracker script, and get a fix and cordinates on the T-beam screen right away, too.
I'm a Mestastic-addict, but I certainly will test out your mapping app !

[image]

kizniche, 2020-08-25:
I don't believe the AXP code is necessary. My example is bare bones and changes the GPS mode to NMEA. Changing baud rate and the resets also appear to be unnecessary.

eriktheV-king, 2020-08-25:
> I don't believe the AXP code is necessary. My example is bare bones and changes the GPS mode to NMEA. Changing baud rate and the resets also appear to be unnecessary.

Indeed, lots of unnecessary stuff. I should have greyed out even more, but I was too much in a hurry to get something working to cheer up the moods.
AXP code needs to wake it up to provide power to ESP32 and GPS, though.

kizniche, 2020-08-25:
> AXP code needs to wake it up to provide power to ESP32 and GPS

The original code I posted works and doesn't have any AXP code.

kizniche, 2020-08-25:
> AXP code needs to wake it up

Indeed you are correct. I suppose my AXP chip was already configured from the ttn-tracker code running and then immediately flashing the test code, without previously unplugging it. I've updated my original code to include minimal AXP code to power the GPS and also the ability to select the board revision so you don't have to mess with pins. I plan to refactor the ttn-tracker code to use this library instead of TinyGPS so this type of reset/reconfiguration will not be necessary.

eriktheV-king, 2020-08-26:
Glad it's all been solved for you, friends!

schroepfer, 2020-09-06:
> > AXP code needs to wake it up to provide power to ESP32 and GPS
> 
> The original code I posted works and doesn't have any AXP code.

Do you implement this code in the ttn-Tracker software?

t0xa, 2020-09-06:
> Do you implement this code in the ttn-Tracker software?

Are you asking about implementing switch back from UBX protocol to NEMA protocol directly in `ttgo-tbeam-ttn-tracker` project?

schroepfer, 2020-09-06:
... exactly.

t0xa, 2020-09-06:
I don't think it's worth it to implement this. It's not directly related to the hardware/software and it's a feature of Meshtastic. Maintainer will need to document, test and support this code. I think it would be enough to just add "Known issues" section in docs with a reset snippet. It's only my view, it's up to @kizniche to decide.

kizniche, 2020-09-09:
I haven't been able to get @geeksville's [modified fork of SparkFun_Ublox_Arduino_Library](https://github.com/meshtastic/Meshtastic-device/issues/334#issuecomment-682045520) to work with my T-Beams. If it were a simple replacement of the TinyGSP library, I may commit to it, but I haven't been able to get the GPS to return coordinates. Additionally, I don't want to add the SparkFun_Ublox_Arduino_Library to this project ***only*** to switch to NMEA mode if it's not already in NMEA mode. It's far easier to add a comment in the Readme for the user to run a small sketch if they have an affected device. If anyone can get the above library to work, I'll see about replacing TinyGPS with it.

peril99, 2020-09-25:
So this has been fun to work thru - but still not seeing results with lat / longitude hitting the TTN mapper app -have a v1.0 device - ran the v3 fixup for nmea - and in the serial monitor is seeing results ... 

[image]

and in the TTN see the messages passing, but not sure how to validate the payload, I see the folloiwng metadata for the message

Payload : AF1C353B4A7500BC0A08
metadata:
{
  "time": "2020-09-25T15:37:52.713835288Z",
  "frequency": 903.9,
  "modulation": "LORA",
  "data_rate": "SF10BW125",
  "coding_rate": "4/5",
  "gateways": [
    {
      "gtw_id": "eui-58a0cbfffe8024ce",
      "timestamp": 684862788,
      "time": "2020-09-25T15:37:52.846798896Z",
      "channel": 0,
      "rssi": -82,
      "snr": 11
    }
  ]
}

and a payload, but not seeing results in ttnmapper, I'm moving the device around and didn't see any different results yet.

kizniche, 2020-09-25:
Did you copy the decoder from step 6 of the Setup in the Readme to your app in TTN?

peril99, 2020-09-25:
> Did you copy the decoder from step 6 of the Setup in the Readme to your app in TTN?

I did not - let me read again. (Jese you are quick! )

peril99, 2020-09-25:
[image]

Cleared it right up - thank you so much - Added a screenshot for other's.

kizniche, 2021-03-24:
Hi. Did you skip [this step](https://github.com/kizniche/ttgo-tbeam-ttn-tracker/issues/20#issuecomment-699016306)?

muckisg, 2021-03-24:
Hi, i copied the decoder function of course if you mean this step.

kizniche, 2021-03-24:
Can you show the output?

muckisg, 2021-03-24:
Here the ttn output:

[image]

right data should be: lat:51.xxxx , long: 7.xxxx

muckisg, 2021-03-24:
Another payload output:

[image]

T-Beam display data is correct but ...

kizniche, 2021-03-24:
Okay. Please start a new issue, as this is different issue from this one.

muckisg, 2021-03-24:
OK

MatthiasHertel21, 2021-05-30:
> We've come to the conclusion that Meshtastic has "broken" the TBeams GPS. We took a know good T-Beam running ttgo-tbeam-ttn-tracker (GPS lock acquired within a few minutes). Then flashed Meshtastic 0.91. That acquired a GPS lock within a few minutes. Then we reflashed ttgo-tbeam-ttn-tracker - now NO GPS.
> We loaded up a basic GPS serial comms sketch and theres no output from the GPS serial port. I tied the battery short reset but that doesnt make a difference.
> I've filed a bug report on the Meshtastic Github page:
> [meshtastic/Meshtastic-device#334](https://github.com/meshtastic/Meshtastic-device/issues/334)
> Just received a shipment of 6 TBeams from LilyGo (all shipped with Meshtastic) and none of them will acquire a GPS lock.

thanks. same behaviour here ...

netmonk, 2021-05-31:
Do you know that meshtastic is modifying gps settings during run ? 
Did you try to reset your gps to factory settings ?

MatthiasHertel21, 2021-05-31:
indeed - it 's working fine after a reset of the GPS Chip ... thanks for your help!

geeksville, 2021-06-06:
To clarify: meshtastic didn't "break" the GPS.  The GPS has configurable protocol options that it stores in its own flash, meshtastic just uses a different set of options (Ublox protocol rather than NMEA) and tells the GPS to make those changes.

robomaniac, 2022-05-08:
> Hello I've modified a LilyGO sketch with SparkFun libraries to help you out to reset U-blox NEO GPS devices on TTGO T-beam T22- V1.0 and 1.1 This sketch will bring back U-blox GPS N6M & N8M factory [...]

this works for me. all my TTGO had meshtactics and could not get GPS to lock on TTGO T-Beam Tracker. I uploaded your code for few minutes then uploaded TTGO T-Beam Tracker again and got everything to work! thanks!

davmu49, 2023-09-02:
Hello,

I have a T-Beam-AXP2101-V1.2 and the gps doesn't works. Meshtastic was already installed on it when i got it

I tried all the sketch above but none of them works for me as they were all written for AXP192.

I'm quite a newbie so i'm unable to correct the code.

Could someone help me please ?

Regards