kizniche/ttgo-tbeam-ttn-tracker #22: Unable to join TTN eith in OTAA or ABP mode !!

vincegre, 2020-08-06:
Hi

I try to use your code to get my TTN T-Beam 1.1 following explanations included but whatever I use OTAA or ABP mode to join TTN network it never shows up on TTN network !
I have edited the file in arduino-lmic library to be setup in 868 (I'm in Switzerland) and also filled my credentials from TTN console (all values in LSB mode) but no way to get it to join TTN network.
I'm in an area covered by TTN (I have already setup some different hardware modules to join TTN network without problems !).

Any ideas what's wrong ?

Thanks

Vincèn

vincegre, 2020-08-19:
No one on this ? have made a complete clean of my Arduino IDE to be sure I'm completely clean of any codes, then reinstalled ESP32 boards, dependencies in Arduino IDE, and then made change in code for EU868 in lmic and added my TTN credentals in ABP mode (used keys in LSB mode unless nothing indicated in documentation or in code) but still unable to get it connect with TTN Lora network :-1:

netmonk, 2020-08-19:
Which board revision do you have ?

kizniche, 2020-08-20:
I haven't been able to reproduce your issue, however I am on the US frequency with my own gateway. Some considerations:

 - Make sure you have the correct board selected in configuration.h, either T_BEAM_V07 or T_BEAM_V10.
 - Does your GPS lock and produce coordinates on screen or through Serial?
 - What are any other messages received by the device when attempting to send or otherwise?

algorni, 2020-08-24:
Just tried on my 1.1 board 
i can see on my gateway the OOTA join request and response but the device still showing "Joining TTN" as it was not able to receive any incoming package to confirm the join.
I remember there was something about wiring some GPIO to being able to enable the data receiving from Lora....  i'll do some research on this.

Then GPS as well, i never got the Lock acquisition even if the red led close to the NEO-6M start to blink....

vincegre, 2020-08-29:
>     * Make sure you have the correct board selected in configuration.h, either T_BEAM_V07 or T_BEAM_V10.

Yep modified to be T_BEAM_V10
 
>     * Does your GPS lock and produce coordinates on screen or through Serial?

Yep GPS communication is fine as it locks as I can see on display as soon as module is outside with clear sky view !
 
>     * What are any other messages received by the device when attempting to send or otherwise?

Not sure what you mean there ?

> Just tried on my 1.1 board
> i can see on my gateway the OOTA join request and response but the device still showing "Joining TTN" as it was not able to receive any incoming package to confirm the join.

I'm not so sure OOTA is supposed to work, I went with ABP mode that is a lot easier to get working as you don't need downlink for it ;)

> I remember there was something about wiring some GPIO to being able to enable the data receiving from Lora.... i'll do some research on this.

It was only on old board generations (no more relevant with v1. or above from what I have read everywhere !

shadow974, 2020-09-22:
same here on my latest v1.1 board. tx power seems to be very low and the lora module is also very deaf. OTAA did only work if a gateway was next to the device. another module is on the way to compare...

maybe swap the lora module to a HopeRF version...

vincegre, 2020-09-22:
I have myself given up with products from TTGO, no support, no uptodate repository with working codes.... and when they answer on their repository it's weeks/months later :(

grahamrobertslx, 2021-01-11:
I too am having this issue. Seems like Rev1.1 is not supported. Any joy in getting it working?

aberridg, 2021-04-26:
I have got my Rev1.1 to work in ABP mode. I cannot get OTA to work at the moment, but I think it's an issue with the underlying MCCA library, or with TTN (I've only tried V3 with OTA so far - going to try V2 when I get the chance). Note that ABP is not LSB first - it's MSB. However, I'm also finding the range to be very poor. Not sure if it's my gateway or the module or my antennas (yet). The board does use the HopeRF module AFAIK.

wally2511, 2021-06-06:
> I have got my Rev1.1 to work in ABP mode. I cannot get OTA to work at the moment, but I think it's an issue with the underlying MCCA library, or with TTN (I've only tried V3 with OTA so far - going [...]

@aberridg I've been trying to get ABP working for a couple of days but still no joy. Any tips? I can see the node connecting to the gateway on OTAA but it seems to be having issues authenticating. On ABP, I don't even see a packet hit the gateway. It's like it isn't even attempting a join. I can get the node to connect and transmit data using the MCCI example ABP sketch.

zoernert, 2022-05-03:
Just a small hint.... I've the same issue. However I've got it working. First of all using the device settings while registering from "Open Source Project" -> "ESP Paxmonitor". Than I double checked the OTAA MSB/LSB coding as at least in my browser it was real hard to read with a overlay if I had the right coding before copy.