kizniche/ttgo-tbeam-ttn-tracker #26: Getting Rev 1.1 working in eu_868

pbgben, 2020-09-20:
I may just be blind, but I could not find out how to set to eu_868, it would always default to us and if you change the build_flags option in the platformio.ini it would just throw errors, so I found the project config and commented out the us_ and uncommented the eu_ definitions. I have also commented out the build_flags line in platformio.ini

Location of the file to change, directory may differ depending on your vscode setup. 
\Documents\platformio\projects\ttgo-tbeam-ttn-tracker-master\.pio\libdeps\esp32\MCCI LoRaWAN LMIC library\project_config

Schrolli91, 2020-10-04:
i have commented out all lines in the LMIC `project_config` file and set the platforIO build flag to eu_868 - works fine for me

grahamrobertslx, 2021-01-11:
I too are having issues with the ver1.1 t-beam board. Managed to get the right frequency but i seem to be having issues getting the NEO-M8 GPS working

netmonk, 2021-01-14:
Commenting everything out in project_config lmic file and replacing the line in platformio.ini with `build_flags = -Wall -Wextra -Wno-missing-field-initializers -O3 -Wl,-Map,.pio/build/esp32/output.map -D CFG_eu868 -D CFG_sx1276_radio` worked fine on rev1.1

RandMental, 2023-05-02:
Any one updated this for the T-Beam V1,.1 with SX1262 LORAS chip?