kizniche/ttgo-tbeam-ttn-tracker #28: Message queued and transmitted but not received on TTN

squeakus, 2020-10-17:
Hey,
First of all thank you for this wonderful code, it is the best documented I have found for the T-Beam and is very cleanly written.

This may be more an issue of my understanding rather of TTN than the code. I have a t-beam 1.0 and a things network gateway. I have setup a new app on it and registered the device EUI  that I saw printed on the serial monitor. I have also added an integration to the application for ttn-mapper.

Regarding the code I have set lmic.h for Europe and set it to use OTAA (which I verified). I have added the app EUI (lsb) to credentials.h and the appkey as copied from the app-> device menu on the TTN console. I have left the DEVEUI set to zero so it automatically populates it. 

It is connecting the to GPS fine and says it is joining TTN, it does not confirm it has joined. It then prints the following:
creen: Latitude:  53.573761

Screen: Longitude:  -6.448909

Screen: Error: 1.53m

Lat: 53.573763
Lng: -6.448909
Alt: 70
Hdop: 15
Sats: 6
Screen: Message queued

TRANSMITTED

I have to assume that even though it says it has transmitted, that nothing has actually been sent across the network. I can see on the gateway it has received 120 messages in the last 3 hours, so it is receiving them, I think!

Are there any steps I have missed in setting up the device or do you have any recommendations for debugging it at this stage? Everything seems to be working but it is breaking down somewhere.

Any help would be greatly appreciated!

squeakus, 2020-10-18:
Actually it turns out I have a v1.1. And some others are having problems connecting to TTN with them:
https://www.thethingsnetwork.org/forum/t/ttgo-t-beam-1-1-success-with-ttn/39340/4

squeakus, 2020-10-18:
I switched to ABP and it immediately started working:
Screen: Message queued

TRANSMITTED
EV_TXCOMPLETE (inc. RX win. wait)
Screen: Message sent

One thing I noticed was the DEVEUI as logged in the gateway was reversed. In the Serial monitor message it was: FF..24 , I added it to the devices list as: FF.. 24 but when I look at the DEVEUI in the gateway it is 24..FF. 

I hard coded it in the credentials file to be LSB 22..FF and while it started picking it up on the gateway, it didn't log any information to the application.

So something very odd is going on with OTA authentication, I suppose I will go with ABP for the moment.

kizniche, 2020-11-26:
I most often us ABP, and I'm not as familiar with OTA. If you happen to find if there is an issue and can resolve it, please share.

J-Paul0815, 2022-04-09:
[https://github.com/kizniche/ttgo-tbeam-ttn-tracker/issues/48](url)
Maybe it will help someone.