kizniche/ttgo-tbeam-ttn-tracker #29: 'LMIC_getSessionKeys' was not declared in this scope

Mohamedx7, 2020-10-21:
I keep getting this message every time i run the code ....any help

/*

Arduino: 1.8.13 (Windows 10), Board: "TTGO LoRa32-OLED V1, 80MHz, 921600, None"

C:\Users\User\Documents\Arduino\main\ttn.ino: In function 'void onEvent(ev_t)':

ttn:160:61: error: 'LMIC_getSessionKeys' was not declared in this scope

         LMIC_getSessionKeys(&netid, &devaddr, nwkKey, artKey);

                                                             ^

C:\Users\User\Documents\Arduino\main\ttn.ino: In function 'bool ttn_setup()':

ttn:251:57: error: 'os_init_ex' was not declared in this scope

     return ( 1 == os_init_ex( (const void *) &lmic_pins ) );

                                                         ^

C:\Users\User\Documents\Arduino\main\ttn.ino: In function 'void ttn_set_cnt()':

ttn:386:26: error: 'LMIC_setSeqnoUp' was not declared in this scope

     LMIC_setSeqnoUp(count);

                          ^

Multiple libraries were found for "SSD1306Wire.h"

 Used: C:\Users\User\Documents\Arduino\libraries\esp8266-oled-ssd1306-master

 Not used: C:\Users\User\Documents\Arduino\libraries\ESP8266_and_ESP32_OLED_driver_for_SSD1306_displays

exit status 1

'LMIC_getSessionKeys' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
*/

closed 2020-10-28 by Mohamedx7

catalinalb, 2021-12-04:
Hello, I have the same issue, this time I choose the verbose option. I am running a fresh install of windows, Arduino IDE and followed carefully all the steps to add the librarys and so on. I also added manually the folders needed for the board (perhaps that might cause an issue? )  here is the error  output: 

Arduino: 1.8.16 (Windows 10), Board: "T-Beam, Disabled, 80MHz, 921600, None"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\catal\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\catal\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\catal\Documents\Arduino\libraries -fqbn=esp32:esp32:t-beam:PSRAM=disabled,FlashFreq=80,UploadSpeed=921600,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10816 -build-path C:\Users\catal\AppData\Local\Temp\arduino_build_57720 -warnings=none -build-cache C:\Users\catal\AppData\Local\Temp\arduino_cache_183756 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\catal\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=C:\Users\catal\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.mkspiffs.path=C:\Users\catal\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\catal\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=C:\Users\catal\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=C:\Users\catal\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -verbose C:\Users\catal\Downloads\ttgo-tbeam-ttn-tracker-1.2.1\ttgo-tbeam-ttn-tracker-1.2.1\main\main.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\catal\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\catal\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\catal\Documents\Arduino\libraries -fqbn=esp32:esp32:t-beam:PSRAM=disabled,FlashFreq=80,UploadSpeed=921600,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10816 -build-path C:\Users\catal\AppData\Local\Temp\arduino_build_57720 -warnings=none -build-cache C:\Users\catal\AppData\Local\Temp\arduino_cache_183756 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\catal\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=C:\Users\catal\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.mkspiffs.path=C:\Users\catal\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\catal\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=C:\Users\catal\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=C:\Users\catal\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -verbose C:\Users\catal\Downloads\ttgo-tbeam-ttn-tracker-1.2.1\ttgo-tbeam-ttn-tracker-1.2.1\main\main.ino

Using board 't-beam' from platform in folder: C:\Users\catal\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6

Using core 'esp32' from platform in folder: C:\Users\catal\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6

cmd /c if exist "C:\\Users\\catal\\Downloads\\ttgo-tbeam-ttn-tracker-1.2.1\\ttgo-tbeam-ttn-tracker-1.2.1\\main\\partitions.csv" copy /y "C:\\Users\\catal\\Downloads\\ttgo-tbeam-ttn-tracker-1.2.1\\ttgo-tbeam-ttn-tracker-1.2.1\\main\\partitions.csv" "C:\\Users\\catal\\AppData\\Local\\Temp\\arduino_build_57720\\partitions.csv"

cmd /c if not exist "C:\\Users\\catal\\AppData\\Local\\Temp\\arduino_build_57720\\partitions.csv" copy "C:\\Users\\catal\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\partitions\\default.csv" "C:\\Users\\catal\\AppData\\Local\\Temp\\arduino_build_57720\\partitions.csv"

        1 file(s) copied.

Detecting libraries used...

"C:\\Users\\catal\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\catal\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/config" "-IC:\\Users\\catal\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/app_trace" "-IC:\\Users\\catal\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/app_update" "-IC:\\Users\\catal\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/asio" "-IC:\\Users\\catal\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/bootloader_support" " [... cut ...] ing" "-IC:\\Users\\catal\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/wpa_supplicant" "-IC:\\Users\\catal\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/xtensa-debug-module" "-IC:\\Users\\catal\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp-face" "-IC:\\Users\\catal\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp32-camera" "-IC:\\Use

catalinalb, 2021-12-04:
OK so I removed the manually added libraries and choose TTGO Lora32-OLED V1. 

With this settings it did compiled and flashed.