kizniche/ttgo-tbeam-ttn-tracker #45: multiple definition of `hal_init';

Raki22, 2022-01-14:
All good until I get to the 7th step of the README file: 

7. Open this project file main/main.ino with the Arduino IDE and upload it to your TTGO T-Beam.

I get this error message: 

_/home/rakilachraf/snap/arduino/61/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/rakilachraf/snap/arduino/61/.arduino15/packages/esp32/hardware/esp32/2.0.2/tools/sdk/esp32/lib/libpp.a(hal_mac.o): in function `hal_init':
(.text.hal_init+0xb4): multiple definition of `hal_init'; libraries/MCCI_LoRaWAN_LMIC_library/hal/hal.cpp.o:/home/rakilachraf/snap/arduino/current/Arduino/libraries/MCCI_LoRaWAN_LMIC_library/src/hal/hal.cpp:416: first defined here
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board T-Beam._

Any help please...thanks in advance.

wolfpcgn, 2022-01-26:
Add:   #define hal_init LMICHAL_init
in lmic_project_config.h

LevendSpacePeak, 2022-03-07:
That worked thanks, btw the correct setting on board setup for me was 40Mhz + 115200 baud  (did not work with default setting 80 + 921600baud )

kizniche, 2022-07-03:
Fixed in https://github.com/kizniche/ttgo-tbeam-ttn-tracker/commit/ae7650100ef6aa566be6e7bff1f4acbdef994fd1

closed 2022-07-03 by kizniche

afifkhoury2005, 2023-12-24:
> Add: #define hal_init LMICHAL_init to lmic_project_config.h
> 
> [mcci-catena/arduino-lmic#714](https://github.com/mcci-catena/arduino-lmic/issues/714)

you made my day, thank you for resolving such issue, I was looking for a solution more than two days.