kizniche/ttgo-tbeam-ttn-tracker #47: Error compiling for board T-Beam

LevendSpacePeak, 2022-03-07:
I have this board : [LILYGO TTGO Meshtastic T-Beam V1.1 CH9102F CHIP ESP32 915Mhz WiFi Bluetooth ESP32 GPS NEO-6M SMA 18650 Battery Holder with OLED (868Mhz OLED)](https://www.amazon.com/LILYGO-TTGO-Meshtastic-Bluetooth-Battery/dp/B08GLFH7J1/ref=sr_1_2?crid=2ODCC0VXEL0FQ)

Currently latest firmware from meshtastic is installed (firmware-tbeam0.7-1.3.3.2fe124e). Device is working fine after firm install, sending messages to phone, can connect to wifi / sats, web server etc. 

Went thru all steps for for second version board, US915 reg. Tried couple times, got errors deploying to board. 
Tried to do again on a new installed system, with freshly installed arduino + libraries (from zip files) .
On Arduino board selection there is "T-Beam" (without TTGO at front) I select it. On menu I see MHZ and baudrate settings, i tried different matchings too. 

Still getting this error even if I just compile or upload: 

> c:/users/pc1/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\PC1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32/lib\libpp.a(hal_mac.o): in function `hal_init':
> (.text.hal_init+0xb4): multiple definition of `hal_init'; libraries\arduino-lmic-master\hal\hal.cpp.o:C:\Users\PC1\Documents\Arduino\libraries\arduino-lmic-master\src\hal/hal.cpp:416: first defined here
> collect2.exe: error: ld returned 1 exit status
> exit status 1
> Error compiling for board T-Beam.

I also have another question about the decoder, I do not see decoder to add on TTN console, I added TTN Mapper webhook integration but still don't see a section to paste the code. I have helium console integration, I added device with same credentials, there is a function section there where I can use decoder code. 

I hope this to work. Thanks for all replies.

wolfpcgn, 2022-03-07:
Just ignore the first error message, it's just an information.
For the second error-massage: read the last thread.

LevendSpacePeak, 2022-03-07:
That worked thanks, btw the correct setting on board setup for me was 40Mhz + 115200 baud (did not work with default setting 80 + 921600baud )

LevendSpacePeak, 2022-03-08:
seems like can not connect TTN also cant get gps lock after 24h outdoor. Credentials correct, I see few helium and TTN spot near my area on coverage maps. Will try 9Dbi lora and another active GPS antenna today.

wolfpcgn, 2022-03-09:
The best thing to do is to get your own gateway.  Than you can be sure that you are not fooled by a single channel or other misconfigurated  gateways.

LevendSpacePeak, 2022-03-09:
Yes but the problem is this unit will be traveling when operating, so should be able to connect any spot it finds. 
I have up to 11 gps tracking but still no luck connecting ttn :( 
Every thing seems correct on cloud side.

wolfpcgn, 2022-03-10:
I would propose to discuss the problem here: https://www.thethingsnetwork.org/forum/t/ttgo-t-beam-topic/15297
because it's not related to kizniches code. I am sure I can help you because I have already 3 T-beam running ABP or OTAA with the code of kizniche.