kizniche/ttgo-tbeam-ttn-tracker #48: Join fails

mdede439, 2022-03-23:
I have a couple of the 1.0 version of the T-Beam modules.  I have been playing around with the code here as a generic sketch and adding additional credential.h folders to hold different credentials for different end devices I create on TTN.  This works, kind of.  I am running in to an issue if I use the credentials to program a device with one set of credentials and then try and reprogram it with another set defined for a different end device.  It appears it retains the old credentials regardless of what I try and overwrite it with in the new sketch.  I was able to fool it and force it to do LMIC_startJoining(); and that seems to write in the new keys and it I see it working on the TTN console.  However, I have two of these boards and I can only get it to work with one of them. Doing this procedure with the second board does write the new keys out to it, because I can see the device address change and I can see the accept join-request and forward join-accept messages but then nothing.  No further packets end up in the end device application.  I see the packets in the gateway from that device address every 20 seconds, as per the sketch, but they don't go where I want them.

Two questions:  Is this code written to program a device just once or is there a facility I have not seen to rewrite the keys in the NVRAM or EEPROM to change the keys?

And what do you think is happening if the end device sends a join and it is accepted but not actually joins?  Any thoughts on where I can look to diagnose this issue?

Thanks for listening and thanks for the code!!

J-Paul0815, 2022-04-09:
I don't know if it helps or if there is a correlation, but this is how I was able to solve my problems, which were already described here:
[https://github.com/kizniche/ttgo-tbeam-ttn-tracker/issues/28](url)
On Screen: Join TTN, Message queued, TRANSMITTED But nothing in TTN, no Join, no Data.
After editing configuration.h

//Uncomment to enable discarding network settings by long pressing second button
#define PREFS_DISCARD

and new flash the T-Beam, I was able to solve my problems by long pressing the middle button after switching it on.
After that everything worked as it should.

mdede439, 2022-04-09:
Thanks for the reply.  That ended up being exactly what I tried and it worked for as well.  I looked more thoroughly through the code and found that option for the the user button to erase the preferences. After implementing  it and giving it a try it worked first time!!  I am still curious what it was keeping saved that even after forcing a rejoin with new credentials that  it would not join.  But like I always say, you can learn more when things don't work than when everything does.