kizniche/ttgo-tbeam-ttn-tracker #52: Request: Only send when location changes

sweetpants, 2022-09-20:
Are there any plans to implement transmitting only when the location changes like:

1) If location changed, transmit in xx min interval
2) If location not changed, transmit every hour or so.

This wil reduce airtime a lot

Regards, Harry

kizniche, 2022-09-20:
No plans for this. Often when testing range it will only infrequently successfully transmit location, therefore every attempt is important, regardless of location or movement.

sweetpants, 2022-09-24:
thnks for the answer, but to save on uplink packets (LoRa-WAN airtime), there is no need to do this if the device has not moved. The only issue i see is when using deepsleep the previous location should be stored in eeprom to be able to calculate distance after waking up from deepsleep. I'm using my tracker in my car, and most of the time it does not move so the last position (in TTNmapper.org) is still valid. I'v already found a way to calculate distance betweem coordinates, now have to implement eeprom functionality to save previous coordinates before going into deepsleep.

kizniche, 2022-09-25:
The main issue I see is the device may assume the coordinates were successfully transmitted to the gateway when they were not. In this case, you would discontinue transmission attempts when you never actually had a point saved to the map.

iw2lsi, 2022-11-09:
what about using on additional I2C/SPI accelerometer instead of the GPS ?
As sweetpants I would like to put the tacker in my car, and to wake it up only when it's moving...

kizniche, 2022-11-16:
There really isn't a question of whether it can be done, but rather should it be done. What's described in this issue are very specific use cases that the majority of users would not need or use, especially if it involves investment in additional hardware.
You're certainly welcome to develop functionality and submit a pull request, but as this kind of feature isn't useful to me, it's not something I will be developing.

iw2lsi, 2022-11-17:
Hi...

   I understand... in the meantime I got my t-beam some days ago and I've successfully tested your firmware on the Helium network... 

It works like a charm.... tnx!

Some days ago I bought also an accelerometer and I'll try to implement the wake-up from sleep feature when a motion is detected on the ESP...  if/when I'll get there I'll publish a pull request so you could decide whether to add this feature or not.

   thanks,

           Giampaolo