kizniche/ttgo-tbeam-ttn-tracker #59: update lilygp t-beam 1.2

RoZelm0, 2023-09-01:
Is it possible to update to this version

https://www.amazon.de/dp/B0B454TB6S?ref=ppx_yo2ov_dt_b_product_details&th=1

RoZelm0, 2023-09-04:
????

mischmeister, 2023-09-13:
The new Lilygo T-Beam uses the AXP2102 PMU, this project currently only support the AXP192 PMU. As far as I know theres an additional library needed as well.

This would be a cool feature.

mischmeister, 2023-09-14:
I don't have much time right now, but this would be the reference to get this done:

https://github.com/lora-aprs/LoRa_APRS_Tracker/commit/b638d951537eb62a43566eb4b02ff6af21a5bc19

The code shows the adaptation of the XPowersLib for use with the AXP2102

RoZelm0, 2023-09-14:
I can try it, but I don't know if I can get it to work.

RoZelm0, 2023-09-16:
I tried it and unfortunately I can't get it to work

garyjduk, 2023-10-30:
I requested this here https://github.com/Xinyuan-LilyGO/tbeam-helium-mapper/tree/feature/web

This does work, it maps and sends to Helium Mapper and Cargo the only thing that isn't quite right is the power management and it says MysteryIRQ for each upload. I think it is getting worked on but this allows you to at least map until it is fixed.

mischmeister, 2023-11-03:
Hello, I recommend you use this project instead:

https://github.com/bitconnector/ttn_mapper_t-beam

He added support for the new chip, you will need platformio to get this running, but the process is really smooth and straightforward

Best regards!